FredsFactory/FreeCAD_AirPlaneDesign #22: Would you mind using git to push commits with more descriptive summaries?

luzpaz, 2020-03-22:
There are many 'Added files via upload' commits in this repo? are you using the github manual upload feature instead of using the `git add` `git push` features from the CLI or from a GUI ?

FredsFactory, 2020-03-22:
Hello
I know, I need to find some time to configure my gui..😊

Fred

luzpaz, 2020-03-22:
@FredsFactory Indeed :)
Check out https://wiki.freecadweb.org/Source_code_management

adrianinsaval, 2020-03-27:
Git kraken is a really good option! Makes it much easier, but you should first familiarize yourself with the terminology and what actually can be done with git. Another practical option is [gitpod](http://www.gitpod.io) wich doesn't require you to install anything and has some gui for git, not as good as gitkraken but useful anyway and better than terminal for some of us.

FredsFactory, 2020-03-27:
Thanks a lot
 I started to read "Git Pro" book and I will try Git kraken .

closed 2020-03-27 by FredsFactory