FredsFactory/FreeCAD_AirPlaneDesign #29: How to Add/Remove Panels in Create A Wing Dialog

rickpalo, 2020-04-07:
In your Create a Wing dialog, how can I add or remove panels?  I can get the example 5 panels, but don't understand how to add or remove a panel.  
Thank you!
[image]

FredsFactory, 2020-04-08:
Hello, I added two buttons one to add row and another to delete existing row. 
Thanks for your comment.
Fred

[image]

closed 2020-04-08 by FredsFactory

rickpalo, 2020-04-08:
That works great!  Thanks for the quick response!!