FredsFactory/FreeCAD_AirPlaneDesign #31: Create Rib allways fails with: 'NoneType' object has no attribute 'extrude'

stumpfwald, 2020-06-07:
Regardless which airfoil I choose (the 2 Epplers provided or another one) the new rib allways has an error.
The Report View shows: 
File "/home/gerhard/.FreeCAD/Mod/AirPlaneDesign/airPlaneRib.py", line 85, in execute
    fp.Shape = face.extrude(FreeCAD.Vector(0,fp.Thickness,0))
<class 'AttributeError'>: 'NoneType' object has no attribute 'extrude'

face returned by process in airPlaneAirFoil.py is allways None.

FredsFactory, 2020-06-07:
Hello,
thank for your issue. Can you confirm the release of freecad that you use ?
I can reproduce the issue on the V0.18 but it works on V0.19.
Fred

stumpfwald, 2020-06-07:
Hello,
I use V0.18.1, the version provided with OpenSUSE 15.1. I could also test V0.18.4 from a community repository with the same result.
Gerhard

FredsFactory, 2020-06-10:
Hello,
I corrected the issue, I'm not being under linux, could you test it?
Fred

FredsFactory, 2020-06-10:
Tested on windows 10 and Mac OS X, on V0.18-16146 and V0.19-21329.

stumpfwald, 2020-06-10:
Hello,
thank you.
Test with V0.18.1 in OpenSUSE 15.1 succeeded.
Gerhard

FredsFactory, 2020-06-10:
Thank for you contribution 
Frédéric

closed 2020-06-10 by FredsFactory