FredsFactory/FreeCAD_AirPlaneDesign #32: Program fails to create a new rib

RAFmaster64, 2020-10-29:
When trying to create a new rib, nothing happens and I get this notice on the bottom stating an error on "airPlaneRib.py", line 243.

[image]

FredsFactory, 2020-10-29:
Hello
thanks for your feed back. 
Some questions :
-can you open the report view (Menu View /Panels/Report vierw)    and copy the error ?
-is it only with NACA profil ?
-did you have the same problem with the 0.19 version ?
Fred

RAFmaster64, 2020-10-29:
Full error:

Running the Python command 'airPlaneDesignWRib' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.18\Mod\FreeCAD_AirPlaneDesign-master\airPlaneRib.py", line 243, in Activated
    editor = SelectObjectUI()
  File "C:\Program Files\FreeCAD 0.18\Mod\FreeCAD_AirPlaneDesign-master\airPlaneDesignProfilUI.py", line 52, in __init__
    self.form=loader.load(path_to_ui)

Unable to open/read ui device

"_-is it only with NACA profil ?_" The error occurs as soon as I select to create a rib, so I don't get to choose the desired profile, if that's what you mean.

I haven't tried with the 0.19 version yet.

FredsFactory, 2021-01-03:
Partially corrected in V0.4.
FreeCAD V0.19 => OK, import file and Naca Generator.

FreeCAD V0.18 under windows :
- Import file => Ok
- NACA generator doesn't work

luzpaz, 2023-03-13:
Still relevant ?

gbroques, 2025-11-29:
I was seeing this same error with FreeCAD 1.0:
```
OS: Ubuntu 24.04.3 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * FreeCAD_AirPlaneDesign 0.4.1
  * CurvedShapesWorkbench 1.0.14
```

closed 2025-11-30 by FredsFactory