FredsFactory/FreeCAD_AirPlaneDesign #35: Module error? also no tool tab in v0.19

Thanitorial, 2021-02-08:
Hi there 
first off great work and thanks for the add on. it realy helps tons. i am using windows to run freecad and have seen you have an issue with the toolbar(wondered how far that is in terms of a fix :) )

i have been trying to get this to work for a bit now and have come up short 
each time i try to load the workbench i get this error
" No module named 'freecad.plot'
  Traceback (most recent call last):
  File "<string>", line 61, in Initialize
  File "C:\Users\Admin\AppData\Roaming\FreeCAD\Mod\AirPlaneDesign\airPlanePanel.py", line 33, in <module>
    from airPlaneRib import WingRib, ViewProviderWingRib
  File "C:\Users\Admin\AppData\Roaming\FreeCAD\Mod\AirPlaneDesign\airPlaneRib.py", line 43, in <module>
    import freecad.plot.Plot as Plot "

i am using the latest version of v0.19 build and could as in v0.18 the addon manager was not working. 
is there a quick fix i can do in the file to ammend this issue or is it a bit bigger than a typo?

thanks loads 
P.S if i am doing something wrong please let me know as i tried to follow instructions set out by yourself and freecad with some success in v0.19

FredsFactory, 2021-02-08:
Hello
thanks for your comment. Simply the workbench need that you install the "Plot" and the "Curved shapes" workbench.
Please say me if it works after test this install 

Fred

Thanitorial, 2021-02-10:
yes that works Thanks for your quick response.

maybe add that to Prerequisites i would have never even have known to install those two if you had not told me about it.
one other question is there a limit to the number of points in the rib? when try 120 it only goes up to 99 (this may seem obvious but still asking anyways)

i look forward to using this tool extensively.
thank you Fred

Thanitorial, 2021-02-10:
if i find anything else i will let you know 
this issue can be closed

FredsFactory, 2021-02-10:
prerequisite updated. 
Thnak for you help
Frederic

closed 2021-02-10 by FredsFactory