FredsFactory/FreeCAD_AirPlaneDesign #36: Install problem Win10

excitedbox, 2021-03-29:
For some reason it says xfoil not found when opening the workspace but the files are being downloaded when using the App Manager. It seems to be a path issue.

I also tried the following 4 methods but with these it doesn't even show up in the workspace list.
I tried manually placing them in both mod folders (APPDATA/Roaming and the FreeCad install directory). And I tried adding them using the Extension manager as well. I also tried adding it as a custom repo in the extension manager.

FredsFactory, 2021-03-30:
Hello,
can you give me the version of freecad that you use ?
Frédéric

excitedbox, 2021-04-04:
The latest win10 build of 0.19 24276. I also have almost all plugins that are compatible with python 3 installed.

I sort of got it running by creating a mod folder in the macro directory and putting it in there, but it is running very buggy. Some features are not showing up or buttons not responding etc.

billdoerr, 2021-09-14:
I am also getting the same issue when navigating to this workbench.  I did have the two prerequisites installed (Plot and Curved Shapes) prior to adding the AirPlaneDesign workspace.

I did try the workarounds submitted by excitedbox, but they had no affect on my install.

15:02:38  No module named 'App.xfoil'
15:02:38  Traceback (most recent call last):
  File "<string>", line 61, in Initialize
  File "C:\Program Files\FreeCAD 0.19\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\bdoerr\AppData\Roaming\FreeCAD\Mod\AirPlaneDesign\airPlanePanel.py", line 33, in <module>
    from airPlaneRib import WingRib, ViewProviderWingRib
  File "C:\Program Files\FreeCAD 0.19\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\bdoerr\AppData\Roaming\FreeCAD\Mod\AirPlaneDesign\airPlaneRib.py", line 38, in <module>
    from App.xfoil.xfoil import XFoil
  File "C:\Program Files\FreeCAD 0.19\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)

15:02:39  No module named 'App.xfoil'

------------
**FreeCAD Info**
------------
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

------------
**System Info**
------------
OS Name	Microsoft Windows 10 Pro
Version	10.0.19043 Build 19043

luzpaz, 2022-04-18:
Is this still an issue?

FredsFactory, 2022-05-11:
yes, after a long absence I will resume the development here in June

NaxiaNoggin, 2022-07-12:
i have this issue too
No module named 'App.xfoil'
they are there in the folders

is there a way to fix this?

im using FreeCad 21, i7, windows

luzpaz, 2022-07-12:
@NaxiaNoggin please add full [about](https://wiki.freecad.org/About) info

NaxiaNoggin, 2022-07-12:
[code]
OS: Windows 10 (10.0)
Word size of FreeCAD: 64-bit
Version: 0.21.29393 (Git)
Build type: Release
Branch: master
Hash: 6820e0a9ec85203a6f342ca72a2ff8fd417beaf1
Python 3.8.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * 3D_Printing_Tools
  * AirPlaneDesign 0.4.0
  * AnimationFreeCAD 1.0.0
  * CADExchanger
  * cura_engine
  * CurvedShapes 1.0.3
  * Curves 0.5.1
  * fcgear 1.0.0
  * LCInterlocking
  * MeshRemodel 1.8919.0
  * Plot 2022.4.17
  * Rocket 2.4.1
  * sheetmetal 0.2.50
  * Ship 2022.4.11
[/code]

NaxiaNoggin, 2022-07-12:
@luzpaz is that the info you mean?

NaxiaNoggin, 2022-07-12:
Processor	Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz, 2592 Mhz, 6 Core(s), 12 Logical Processor(s)
System Model	Predator PH315-53
OS Name	Microsoft Windows 11 Home

davesrocketshop, 2023-01-30:
There is a name conflict with the App folder. If I rename it to something else, like 'Stuff' (not a good name I know), it works.

I'm not sure where the conflict originates. I use App in my workbench, but I couldn't use Gui as it conflicted with the core folder name. I'll try investigating more, but renaming the folder is a usable work around

luzpaz, 2023-02-05:
CC @FredsFactory