FredsFactory/FreeCAD_AirPlaneDesign #45: No Module name App.efoil

mmwaileo, 2022-10-02:
Hi, I just install AirPlaneDesign Workbench. when i selected the workbench, receive this error message.
[image]
anyone know how to solve this problem. i am working on windows 11, Freecad 0.20.1
here are errors show on report view:
No module named 'App.xfoil'
Traceback (most recent call last):
  File "<string>", line 61, in Initialize
  File "C:\Program Files\FreeCAD 0.20\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\DELL\AppData\Roaming\FreeCAD\Mod\AirPlaneDesign\.\airPlanePanel.py", line 33, in <module>
    from airPlaneRib import WingRib, ViewProviderWingRib
  File "C:\Program Files\FreeCAD 0.20\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\DELL\AppData\Roaming\FreeCAD\Mod\AirPlaneDesign\.\airPlaneRib.py", line 38, in <module>
    from App.xfoil.xfoil import XFoil
  File "C:\Program Files\FreeCAD 0.20\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)

eric-the-eye, 2022-10-30:
Rocket workbench causes AirPlaneDesign workbench to fail

davesrocketshop, 2023-03-13:
Yes, and while I believe the installation of both applications is the cause, both are at fault. Much of the documentation describes App and Gui folders for workbenches and this is the core problem as it runs into naming conflicts. Early on I discovered the Gui conflict and renamed my package as Ui. The App folder showed no issues so I left it as is.

I tried renaming the App folder in the airplane design workbench and things worked fine. It's a straight up name space problem. I'll rename my folder in my next release, and the airplane design workbench should probably do the same.

davesrocketshop, 2023-03-14:
Unfortunately renaming my App folder is a no go. Unfortunately when importing older models there is no way to migrate the old class and path to the new class and path without the old path existing. Which of course defeats the purpose. I'll raise this on the forum but I don't believe this can be fixed.

https://wiki.freecad.org/Scripted_objects_migration

davesrocketshop, 2023-03-14:
And once again the forums pull through! https://forum.freecad.org/viewtopic.php?p=667207#p667207

I will be renaming my App folder in the next release. This workbench should do the same.

I've tested with my renamed folder and this workbench now works as expected.